openpower-cores/a2i #1: Code not compiling

LarsAsplund, 2020-06-30:
The provided code fails to compile (with Modelsim) due to

1. Declarations are missing for a number of attributes. For example `like_builtin` used [here](https://github.com/openpower-cores/a2i/blob/7d9acfcad3456cd166224e60deb6d8583b6fa560/rel/src/vhdl/ibm/std_ulogic_support.vhdl#L51).

2. The references to libraries `latches` and `macros`. They are not used so the code compiles without them. [Here](https://github.com/openpower-cores/a2i/blob/7d9acfcad3456cd166224e60deb6d8583b6fa560/rel/build/ip_user/a2x_axi/a2x_axi/src/work/xuq_lsu_cmp_cmp31.vhdl#L25) is an example.

LarsAsplund, 2020-06-30:
More specifically the following attribute declarations are missing:

like_builtin
functionality
dc_allow
type_convert
btr_name
pin_bit_information
recursive_synthesis
dynamic_block_data
block_data

openpower-cores, 2020-07-15:
Made some changes for the above problems - can you try compiling again?  Thanks...

xinyu8888, 2020-08-11:
> More specifically the following attribute declarations are missing:
> 
> like_builtin
> functionality
> dc_allow
> type_convert
> btr_name
> pin_bit_information
> recursive_synthesis
> dynamic_block_data
> block_data
@LarsAsplund Hi buddy! Have you already solved the problems you mentioned above? If the answer is yes, how did you do that? We are currently encountering some similar issues.

LarsAsplund, 2021-04-02:
@openpower-cores The issues I had with ModelSim are no longer there. Thanks

closed 2021-04-02 by LarsAsplund