openpower-cores/a2i #6: How could i simulaion it in vivado or any other tools?

tyskink, 2020-08-07:
in vivado, it will report many errors focus in the ibm library. e.g. boolean not match integer literal, the invert function has too many element

what's more, i tried irun 15.2 and VCS Mx, they all reported too many compile errors.

how do you run the simulation and verification? how could i run a high level simluation?

thanks

openpowerwtf, 2020-08-07:
Do you mean behavioral or netlist simulation?
We ran a small amount of both, but mainly used non-Vivado cycle simulation at core level.

tyskink, 2020-08-10:
Thanks very much. 
But I still not very clear about how to run a simulation. Do you have any tutorial?

openpowerwtf, 2020-08-10:
I can look for any tcl fragments I may have, but there won't be much.  Some background:

When I got the source and libs dusted off and compiled, I immediately hit a problem - xelab segfaulted with no stack trace.  I couldn't narrow it down by hacking up the logic, except to know it was caused in xu.  It was probably three months before Xilinx sent a patch.  So there was no sim done in Vivado during that time, and we were already working on implementation fails by the time xelab was working.  I did a very small amount of Vivado behavioral and netlist sim after that.

If using the board design, you should not have to do much except wave the diff clk inputs.  When reset goes inactive,  you should see core activity:
* ierat/derat por_seq_q run through some steps to load the erats with boot entries
* core fetch to FFFFFFFC (requires threadstop(0)=0; I think it inits to '1' from VIO)
* axi_reg returns 48000002 op
* core fetch to 00000000
* if you haven't loaded bram, core fetch to 000000E0 repeatedly

xinyu8888, 2020-08-11:
@openpowerwtf Hi bro! I just read your comments above and they are really helpful for us. We are currently doing FPGA on-board design with A2 core. Could you please pinpoint some more typical signals related to the core logic that we can capture on ILA to help us better understand the core? Thank u in advance.

xinyu8888, 2020-08-11:
@openpowerwtf Besides, the open source vhdl code is really not that easy to analyze and interpret. There is zero comment and so many acronyms in the signal name that we cannot get any reference with in the mannual book. Is there any other document that we can refer to so that we can better understand the code? Or can we expect to have a commented version of verilog code being released in the near future? Thx!

openpowerwtf, 2020-08-11:
@xinyu8888 

Yes, we are working on 'reinstating' the relevant comments!  They will help, especially around the obvious generated code.  Should be soon.  

There are some remnants of useful signals in ila_axi.tcl.  But we didn't try gen'ing with 'dont-touch' to ensure nets stayed around after implementation.  I think it would time OK like that if you want to try.  The very basics you want are the A2L2 interface and things like the NIA and GPR write valids/data and various pipe stage valids.   Until you have code that sets up translation (ERAT-only or TLB), all accesses are I=1 (cache-inhibited), so you will see a lot ifetches/data reads on the bus, and it is store-through so you see all writes.

We didn't get far enough to get the 'scancom' interface up and working.  That is the serial interface for status/control to/from the core and allows on-the-fly access to a lot of the core regs.  There are also LOTS of debug bus options documented.

I assume you have looked at the user guide in the doc.  There are some useful diagrams of pipe and functional units.  And lots of acronyms!  Maybe we could do something with the Wiki to help there.

xinyu8888, 2020-08-11:
@openpowerwtf Thanks a lot bro! Excellent comments! Really appreciate it! Our team is carrying out deep research on A2 core. I might need to bother you a lot on this page in the following days. Please let me know if your team has any new updates related to the code comments released in the near future. Thanks again!

openpowerwtf, 2020-08-29:
The source code has been updated with comments.

closed 2020-08-29 by openpowerwtf