openpower-cores/a2i #17: The speculative excution of instructions relying on previous loads

Grubby-CPU, 2020-09-01:
In the A2I manual, it says A2I supports speculative issuing instructions which are dependent on a load based on the assumption that the load will hit in the data cache. I remeber this technique is also designed in Alpha21264, cool!

My quesiton is if I want to learn more details about this technique in A2I, which files and what key signals should I pay attention to? In the iuq_fxu_dep.vhdl, there are some codes related to "shadow pipeline", are these related to this speculative technique?

openpowerwtf, 2020-09-02:
This is similar to #16.
Yes, that logic is obviously central to the fx pipeline tracking.  For specifically the ld-dep case, seems like cmiss/target reg are the key things, since that turns it into a flush/stall case. 

I don't know what the 'shadow pipeline' officially refers to.  I though it might be this, but the stages don't match:

> A flush is triggered at one of five stages in the pipeline: IU2, IU5, RF0, EX1, or EX4.

But the term is used in other units too.  It may be the 'control' parts of the pipelines for tracking valids, checking dependencies, etc. (vs. the datapaths).

closed 2020-09-09 by openpowerwtf

Grubby-CPU, 2021-08-30:
Shadownpipeline is used to support the single instruction mode. If one instruction exits in the pipeline and a single instruction mode is enabled, dependency happens and the next instruction is stalled until the current instruction totally finishes.