openpower-cores/a2i #18: Questions about MMQ TLB xbit hole

Huash107, 2020-09-01:
Hi, guys. I'm looking the memory xbit part.

If the xbit is used in a TLB entry and the corresponding effective address, how does it match?

As what I know, the matching process does not distinguish the "hole" or the outside part.

The process just identify the page size part of TLB entry and address-in,(0 to N, N = 64 - log2(page-size),
 
then match the  N to 63 part of entry and address-in.

Does anyone have the same or different thoughts?

openpowerwtf, 2020-09-01:
It looks like mmq_tlb_matchline.vhdl does consider x-bit (if generic is set to enable it).  Do you think it's not implemented as described in **6.2.3 Exclusion Range (X-bit) Operation**?

> When the X bit of a translation entry is set, a subset of the entry’s EPN least significant bits can be set
> to one to define an address match exclusion range such that any effective address low enough in the
> page to fall into the exclusion range will not compare. This allows for
> large pages to be defined that possess a subregion that can be filled in with smaller page sizes, which would
> normally overlap (coexist within) the larger page. This allows for more efficient use of entries (especially in
> ERAT-only mode) because the user does not need to “fill in” with small pages up to a larger page alignment
> in the system memory map (potentially using more entries with this approach). It also provides for more flexibility
> in mixing large and small page organization in the overall system memory map.

Huash107, 2020-09-01:
Thank you for answering my question. But I'm still a little confused about the TLB EPN to RPN.

As I see the process of transmission in the mmq_tlb_matchline
After the identification of pgsize, the "function_x_51" part is used as follow:
```
 function_34_51 <= not(entry_xbit) or
                    not(pgsize_eq_1G) or
                    or_reduce(entry_epn_b(34 to 51) and addr_in(34 to 51));
  -- 256M
  function_36_51 <= not(entry_xbit) or
                    not(pgsize_eq_256M) or
                    or_reduce(entry_epn_b(36 to 51) and addr_in(36 to 51));
  -- 16M
  function_40_51 <= not(entry_xbit) or
                    not(pgsize_eq_16M) or
                    or_reduce(entry_epn_b(40 to 51) and addr_in(40 to 51));
  -- 1M
  function_44_51 <= not(entry_xbit) or
                    not(pgsize_eq_1M) or
                    or_reduce(entry_epn_b(44 to 51) and addr_in(44 to 51));
  -- 64K
  function_48_51 <= not(entry_xbit) or
                    not(pgsize_eq_64K) or
                    or_reduce(entry_epn_b(48 to 51) and addr_in(48 to 51));
```
is that means the RPN(real page number is a part of EPN?
just EPN(0-51)  + offset is the real address?

Huash107, 2020-09-01:
Thanks for your answering. But I still have 2 question that expecting you to help. 

Q1 . If a 1MB page has a 64KB "hole“, does it need 2 TLB entries to map the virtual address to the 1MB physical address?

Q2 . If we need 2 entries according to Q1, then what are the pagesize of the two entries?

Thank you very much!

luffygood, 2020-09-01:
These are some signals in the mmq_tlb_matchline.
```
  comp_or_48_51 <=  and_reduce(match_line(48 to 51)) or pgsize_gte_64K;
  comp_or_44_47 <=  and_reduce(match_line(44 to 47)) or pgsize_gte_1M;
  comp_or_40_43 <=  and_reduce(match_line(40 to 43)) or pgsize_gte_16M;
  comp_or_36_39 <=  and_reduce(match_line(36 to 39)) or pgsize_gte_256M;
  comp_or_34_35 <=  and_reduce(match_line(34 to 35)) or pgsize_gte_1G; -- glorp
```
I really want to know what do these signals do when matching the TLB entry with the effective/ virtual address. Want some help.

openpowerwtf, 2020-09-01:
@Huash107 Depends why you are creating a hole.

**6.2.3 Exclusion Range (X-bit) Operation**
> The rules for configuring an exclusion range “hole” for a given TLB entry and placing one or more pages
> within the “hole” are as follows:
> 1. Only TLB entries with page sizes greater than 4 KB can have an exclusion range hole enabled via X = 1.
> 2. A virtual address to be translated that falls within the hole will not match this TLB entry.
> 3. The size of the hole configured must be smaller than the page size of this TLB entry.
> 4. The size of the hole is configurable to 2<sup>n</sup> x 4 KB, where n = 0 to log<sub>2</sub> (entry size in bytes) - 13.
> 5. The legal binary values of the unused EPN bits of a given TLB entry are contained in the set defined by
> 2<sup>n</sup> - 1, where n = 0 to log<sub>2</sub> (entry size in bytes) - 13.
> 6. Other TLB entries of valid page sizes (less than or equal to the hole size) can be mapped into the hole.
> 7. Multiple other TLB entries can be mapped into the hole simultaneously.
> 8. Not all of the address space defined by the hole needs to be mapped by other entries.
> 9. Pages mapped in the hole must be page-size aligned.
> 10. Pages mapped in the hole must not overlap.
> 11. Pages mapped in the hole must be collectively fully contained within the hole.

Huash107, 2020-09-01:
I'm still confused about the xbit transmission. Could you show me a example with detail?

closed 2020-09-03 by openpowerwtf