openpower-cores/a2i #21: The 'pgsize_eq' of the TLB entries with xbit

Huash107, 2020-09-02:
According to the BGQ article,  If a 1M page have 64KB hole, what are the 'pgsize_eq' of the remaining 960KB TLB entry and the pagesize of the 64KB. 

I think for the 960KB page, its 'pgsize_eq' is 1MB  and the 64KB hole has 64KB 'pgsize_eq'

Is it right?

openpowerwtf, 2020-09-02:
That is one legal possibility.

**6.2.3 Exclusion Range (X-bit) Operation**
> The rules for configuring an exclusion range “hole” for a given TLB entry and placing one or more pages
> within the “hole” are as follows:
> 1. Only TLB entries with page sizes greater than 4 KB can have an exclusion range hole enabled via X = 1.
> 2. A virtual address to be translated that falls within the hole will not match this TLB entry.
> 3. The size of the hole configured must be smaller than the page size of this TLB entry.
> 4. The size of the hole is configurable to 2<sup>n</sup> x 4 KB, where n = 0 to log<sub>2</sub> (entry size in bytes) - 13.
> 5. The legal binary values of the unused EPN bits of a given TLB entry are contained in the set defined by
> 2<sup>n</sup> - 1, where n = 0 to log<sub>2</sub> (entry size in bytes) - 13.
> 6. Other TLB entries of valid page sizes (less than or equal to the hole size) can be mapped into the hole.
> 7. Multiple other TLB entries can be mapped into the hole simultaneously.
> 8. Not all of the address space defined by the hole needs to be mapped by other entries.
> 9. Pages mapped in the hole must be page-size aligned.
> 10. Pages mapped in the hole must not overlap.
> 11. Pages mapped in the hole must be collectively fully contained within the hole.

> For example, for a 64 KB page size, the EPN bits 48:51 is ordinarily set to zero because they do not participate
> in the virtual address matching. The set of legal values for EPN(48:51) representing the hole size would
> be {0000, 0001, 0011, 0111}. If software needs to create a 16 KB exclusion range at the base of the 64 KB
> page, it sets X = 1 and the EPN bits 48:49 = ‘00’ and 50:51 = ‘11’ (EPN bit 50 is the MSB of the 16 KB
> address base to exclude). Addresses above the first 16 KB, but still within the 64 KB page, match. Addresses
> within the first 16 KB of the 64 KB do not match. Software is then free to create four additional 4 KB virtual
> pages overlaying the 16 KB hole within the 64 KB page in question.

Huash107, 2020-09-02:
Thanks for your answer! According to the 6.2.3, the 64KB has pgsize_eq=64K, which means its 48 to 51bit are set to be 0000, And the 980KB in 1M page are set to be 0001 0000 from 44 to 51bit. 

So is there any other possible for this example? If it is, could you give me some other example please?

Huash107, 2020-09-02:
Is the have_xbit of each effective address various?  For the real space mapped to effective page with hole, its have_xbits is 1 while the have_xbit is 0 for those EA without hole.

openpowerwtf, 2020-09-02:
> Thanks for your answer! According to the 6.2.3, the 64KB has pgsize_eq=64K, which means its 48 to 51bit are set to be 0000, And the 980KB in 1M page are set to be 0001 0000 from 44 to 51bit.
> 
> So is there any other possible for this example? If it is, could you give me some other example please?

Yes, if you are asking about your original example (1M w/64K hole).  Items 6-8 state that all combos of legal pages (including none) can be mapped to a hole, following the placement rules in items 9-11.

closed 2020-09-09 by openpowerwtf