openpower-cores/a2i #25: How to quickly find the key information of the combination logics

Grubby-CPU, 2020-09-07:
Hi Guys,

  In A2I, there are so many combination logics. For example, in the xuq_add_loc.vhdl file, there are many "and", "or", "not" gates as shown below.

`   

    u_g01t  : g01_t  (0 to 7) <= not g01_b(0 to 7) ; 
    u_g01not: g01_not(0 to 7) <= not g01_t(0 to 7) ; 
    u_z01b:   z01_b  (0 to 7) <= not t01_b(0 to 7) ;
    u_p01b:   p01_b  (0 to 7) <= not( g01_not(0 to 7) and z01_b(0 to 7) );
    u_p01:    p01    (0 to 7) <= not( p01_b(0 to 7) );

    u_g08i_0: g08_b(0) <= not g08(0) ;
    u_g08i_1: g08_b(1) <= not g08(1) ;
    u_g08i_2: g08_b(2) <= not g08(2) ;
    u_g08i_3: g08_b(3) <= not g08(3) ;
    u_g08i_4: g08_b(4) <= not g08(4) ;
    u_g08i_5: g08_b(5) <= not g08(5) ;
    u_g08i_6: g08_b(6) <= not g08(6) ;
    u_g08i_7: g08_b(7) <= not g08(7) ;

    u_t08i_0: t08_b(0) <= not t08(0) ;
    u_t08i_1: t08_b(1) <= not t08(1) ;
    u_t08i_2: t08_b(2) <= not t08(2) ;
    u_t08i_3: t08_b(3) <= not t08(3) ;
    u_t08i_4: t08_b(4) <= not t08(4) ;
    u_t08i_5: t08_b(5) <= not t08(5) ;
    u_t08i_6: t08_b(6) <= not t08(6) ;
    u_t08i_7: t08_b(7) <= not t08(7) ;

    u_sum_0_0: sum_0(0) <= not(  ( p01(0) and  g08(1) ) or  ( p01_b(0) and  g08_b(1) )   ); --output--
    u_sum_0_1: sum_0(1) <= not(  ( p01(1) and  g08(2) ) or  ( p01_b(1) and  g08_b(2) )   ); --output--
    u_sum_0_2: sum_0(2) <= not(  ( p01(2) and  g08(3) ) or  ( p01_b(2) and  g08_b(3) )   ); --output--
    u_sum_0_3: sum_0(3) <= not(  ( p01(3) and  g08(4) ) or  ( p01_b(3) and  g08_b(4) )   ); --output--
    u_sum_0_4: sum_0(4) <= not(  ( p01(4) and  g08(5) ) or  ( p01_b(4) and  g08_b(5) )   ); --output--
    u_sum_0_5: sum_0(5) <= not(  ( p01(5) and  g08(6) ) or  ( p01_b(5) and  g08_b(6) )   ); --output--
    u_sum_0_6: sum_0(6) <= not(  ( p01(6) and  g08(7) ) or  ( p01_b(6) and  g08_b(7) )   ); --output--
`
  I know this file is related to the ALU add operation but I have no idea about how it works. Any idea about how to read these codes?

Many thanks

openpowerwtf, 2020-09-08:
I would assume that logic that is very low-level like this was copied/updated from a previous version, and/or written specifically for timing/PD reasons.  It's essentially a 'netlist' version, to be synthesized 1:1; something like a hard macro, but not quite .  This appears to be implementing some piece of the adder without XOR's, so this was likely critical logic; maybe it had slightly better performance vs. a synthesized version, and got repeatable results across synthesis runs by controlling cells/fanouts/etc.  

If you want to understand it, you'll have to figure out the combinational functions and rewrite in a more human-readable form.

Grubby-CPU, 2020-09-08:
Hi openpowerwtf，
  Thanks! The remaining question is how to figure out the combinational functions:)) Any suggestions or recommended tools?

openpowerwtf, 2020-09-09:
No, it's ugly when the logic is written like that.  You can try to reconstruct it bottom-up, but it may not be obvious what it's doing even if you turn it into higher-level equations.  I would start top top-down and see where the macros fit in.   If you can guess something is a CLA, CSA, etc.,  or a piece of one, you can check some of the logic inside to see if it makes sense.  Then if you really want to verify it, do a simple testbench with just the macro and see if the function matches what you thought.

closed 2020-09-12 by openpowerwtf