openpower-cores/a2i #26: L1 cache interface

darwin964, 2020-09-08:
Hi，in order to implement L2 cache,  we want to make an block level testbench contain L1 and L2 cache，how could I get the L1 interface? or get L1 rtl block?

openpowerwtf, 2020-09-08:
The L1 I/D caches are part of the core rtl.

This is the spec for the core interface:
[A2L2 Spec](https://github.com/openpower-cores/a2i/blob/master/rel/doc/A2L2.pdf)

darwin964, 2020-09-09:
@openpowerwtf Is A2 designed by full custom?

openpowerwtf, 2020-09-09:
No, I don't think any of the random logic was custom.

openpowerwtf, 2020-09-10:
Don't forget  - BlueGene ran at a slow frequency to balance system power/perf.  Would have to synthesize to real technology today to see what the critical paths are; very likely some are around arrays.

closed 2020-09-12 by openpowerwtf