openpower-cores/a2i #32: Problem in simulation of the a2l2_axi.vhdl file

xinyu8888, 2020-10-28:
Hi buddy @openpowerwtf , we are currently doing simulation of the a2l2_axi.vhdl file. We first initialized the cache by storing random data into all addr consecutively. After all storing commands finished, we first issued load (001000) and iftech (000000) req instructions in two consecutive cycles to fetch 4 words. The an_ac_reld_data we got is correct in this case. Then we issued load (001000) and ifetch (000000) req instructions in two consecutive cycles again to fetch 16 words, we noticed for the first group of 16 words, the last 4 words were missing as they were not fetched into an_ac_reld_data (all zero) because rld_data_qw3 is all zero. But the second group of 16 words were all fetched into an_ac_reld_data properly. What might be the cause of missing the last 4 words? Is this a bug? Thanks!
[image]

openpowerwtf, 2020-10-29:
Are you doing the sequence with instructions (not sim driver code)?  If you are mixing 16B and 64B loads, you must be using I=1 and I=0 (cmiss) accesses.  Are you using the current version of a2l2_axi?  I did put some changes in a while ago while I was using it to bring up A2O.

xinyu8888, 2020-10-29:
@openpowerwtf , We were simply giving ifetch instructions to the ac_an_req_ttype input signal. Yes, we used I=1 and I=0 accesses for mixing 16B and 64B loads. The a2l2_axi version that we are currently using is the newest. And we just found another bug in this file. I will upload the issue shortly.

xinyu8888, 2020-10-29:
@openpowerwtf After we initialized the cache by storing random data into all addr consecutively, we sent a bunch of ifetch instructions to the ac_an_req_ttype input port. Among all the 200 sets of 16B data we got from ifetch, one set reported error due to missing the last 4 bytes in the an_ac_reld_data output port as indicated in the error log file and the screenshots (I uploaded the waveform screenshot of both a bad one and a good one).
[image]
[image]
[image]

openpowerwtf, 2020-11-04:
Sounds like a bug - definitely hasn't been much testing with I=0/I=1.  Or maybe something unusual happening on AXI?   Any error bits active in A2L2_AXI?

xinyu8888, 2020-11-05:
@openpowerwtf Axi looked good all the time. We didn’t see any error bits active in A2L2_AXI. Does your team plan to do more testing with a2l2_axi.vhdl and eventually fix all the potential bugs? If yes, may I know the schedule of your plan? We found some other issues that we haven't uploaded yet.  We need to figure out how we should proceed with our L2 cache design work based on your plan. Thx.