openpower-cores/a2i #34: Question regarding the address map of A2i core

xinyu8888, 2020-11-03:
Hi @openpowerwtf , does A2i core have an address map that is similar to the one mentioned in the attachment? When we first allocated 512M address space to DDR memory starting at address 0x8000_0000, DDR did not work properly. We were able to write data to DDR and read correct data from it through JTAG, but when the core started running, we didn’t see any data being transferred through the axi bus between the core and DDR. Then we changed the starting address of DDR to 0x2000_0000, the core and DDR started working properly and we saw data being transferred through the axi bus between the core and DDR. So we are wondering if there is also an address map of A2i core for certain peripherals? For example, if we want to add DDR, SPI bus connected to SD card and UART as peripherals for A2i core, what would be the ideal starting address for them? Thx!
[image]

xinyu8888, 2020-11-03:
@openpowerwtf The same goes for UART. When we first set the starting address of axi_uart to 0x4060_0000, we always had SLVERR on axi_bresp signal. After we changed the starting address of axi_uart to 0xFFFC_0000, everything worked fine.

openpowerwtf, 2020-11-04:
The AXI address map is set up in the project, for accessing the addressable macros.  You should be OK if you set up everything consistently (sounds like it, since JTAG-AXI works). 

Have you set up the translation correctly for new address space(s)?  You need to map those spaces through ERAT (ERAT-only) or page table.   You will get a DSI or a bad address if something's wrong there.

zhaoxiahust, 2020-11-09:
To access the 512M address space, I guess it requires a simple OS that can add the TLB entries to do the VA to RA translation when the TLB miss exception happens.

openpowerwtf, 2020-11-09:
You can use TLB, or you can add entries for ERAT-only addressing, just like the simple kernel is doing for several ranges.  And the core logic initializes two entries after reset (section 4.2 of UM), which you could change for your purposes.

zhaoxiahust, 2020-11-09:
Hi openpowerwtf, if I want to add some TLB-related functions such as adding the TLB entry if the TLB miss happens during the execution, what should I do? Can you give me some hints about this?
I guess I can add some TLB-related functions in the boot.s file and it begins to work when the core meets the TLB miss exception. Is it the right direction?

openpowerwtf, 2020-11-09:
It is much easier to start by using ERAT-only mode and set it up so your effective addresses always hit the ERATs, so you won't have translation misses to handle.  Sections 6.14/6.10 describe that mode.  The biggest page size is 1G so you won't need many entries, unless you want to set up different translations/permissions/etc.

zhaoxiahust, 2020-11-09:
Yes, I totally understand this:). 
I am just curious how does OS handle TLB miss exception in IBM world and whether we can implement a simple version to achieve this in the boot.S file....

openpowerwtf, 2020-11-09:
Yes, it's not too hard to implement a basic page table handler - you just have to understand the details.  I have written one for a 'bare-metal' test environment.  FYI - the translation mechanism has changed to 'radix tree' style in newer versions of the ISA.

closed 2021-01-05 by openpower-cores