openpower-cores/a2i #44: Why choosing ex4 to deal with interrupt and exception

Grubby-CPU, 2021-10-12:
As shown in xuq_cpl.vhdl, ex4 is used to deal with the interrupt and exception. Why ex4 instead of ex3 or ex5? Any comments about this?

zhaoxiahust, 2021-11-13:
I think there are many reasons related to this. First, you want to move this function as early as possible since it flushes the whole pipeline which degrades the performance. Second, you need to put this function behind some events such as injecting a load miss into the LMQ, getting the right branch prediction result and so on.

Let's look forward to the detailed explanation from @openpowerwtf .

openpowerwtf, 2022-03-03:
Have you read Appendix D of the spec?  There are lots of details about the pipe n general, and arbitration, stalls, and flushes.