openpower-cores/a2i #51: About some unknow instructions in A2 Core

Grubby-CPU, 2022-06-08:
Hi all,

  When I was reading the A2 code, I found some unknown codes (xuq_dec_b.vhdl) as shown below. I guess these instructions are related to DITC (direct interthread communication) but I cannot find any information in the A2 manual or the power ISA manual. Any help?

```
rf1_is_mfdp                  <=  '1' when rf1_opcode_is_31_q(3) = '1'  and   rf1_instr_21to30_04_q(21 to 30)    = "0000100011"                              else '0';
rf1_is_mfdpx                 <=  '1' when rf1_opcode_is_31_q(3) = '1'  and   rf1_instr_21to30_04_q(21 to 30)    = "0000000011"                              else '0';
rf1_is_mtdp                  <=  '1' when rf1_opcode_is_31_q(3) = '1'  and   rf1_instr_21to30_04_q(21 to 30)    = "0001100011"                              else '0';
rf1_is_mtdpx                 <=  '1' when rf1_opcode_is_31_q(4) = '1'  and   rf1_instr_21to30_04_q(21 to 30)    = "0001000011"                              else '0';

```

openpowerwtf, 2022-06-14:
I am not sure if DITC was included in any ISA level.  I can try to find some documentation.  I only see it described in the A2L2 spec relative to its bus activity. 

I do see it referenced in several patents from that timeframe; e.g. https://patents.google.com/patent/US8640230.

openpowerwtf, 2022-06-14:
It is briefly mentioned in the A2O user manual, but the function is also not described further; there are some related SPR bits.  It may have been an 'experimental' feature.