openpower-cores/a2i #52: Address space of an SoC using A2I core

Grubby-CPU, 2022-06-29:
Hi,
  I am curious about the memory address space of an SoC such as Blue Gene/Q using the A2I core. Besides this, I also tried to search the address space of the powerPC and power-series server but also did not find any useful information. Any recommended references?

openpowerwtf, 2022-06-29:
What do you mean by 'address space'?  Do you mean address translation, or how software uses the memory space?

Grubby-CPU, 2022-07-01:
I mean in Power servers, which memory space is assigned to DDR and which memory space is assigned to I/O devices. For example, 0 - 2GB is assigned to I/O, 2GB-4GB is assigned to DRAM, 4GB-16GB is reserved and 16GB-32 GB is assigned to I/O and so on. Any public references?